pschatzmann/arduino-stk #2: No mentions of the dependencies (ESP32-A2DP)

MrKOSMOS, 2021-11-27:
Hello, this is an amazing piece of software, and after 1 year it finally seems to to work on my machine! Sadly, I got compilation errors at first, and only after checking the error output did I realize that it needed a .h file included in your ESP32-A2DP library to make it work. Which it did afterwards.

Now this seems to call for one of three solutions:

- Remove from Arduino-STK anything that needs or mentions ESP32-A2DP, as long as that doesn't remove any functionality;
- Include any necessary .h file from ESP32-A2DP in Arduino-STK
- Mention ESP32-A2DP as a dependency in the description/installation notes

Anyways, this project is lovely. I was struggling making a synth library on my own in Arduino, and even considered using the ESP-IDF, but now I can finally (mostly) rest my head. Not even Mozzi has this much functionality on the ESP32, and it's one of the most overused libraries out there...

P.S. Also if you could make an "include all" file for the Arduino IDE to import, instead of it placing includes for every single header file in front of the code, that would be pretty cool.

pschatzmann, 2021-11-28:
Thanks for your feedback. I changed the ArdConfig.h and commented the A2DP functionality out, so that the library is not necessary any more. I also updated the Readme.

I was missing the "include all" as well. So I have create the new StkAll.h

MrKOSMOS, 2021-11-28:
Cool! I guess it's done now

closed 2021-11-28 by MrKOSMOS