pschatzmann/arduino-stk #3: i2s

bazworkshops, 2021-12-25:
Trying to use i2s. The old clarinet-i2s Example works through audio tools, but I can't get sound out in the new InstrumentToI2S one. Got the error: ArdI2SOut doesn't have "SetPins", so I first commented that line and put my pin numbers in ArdI2SOut.h as defaults. Nothing. I figured to change "SetPins" in InstrumentToI2S to "setPinConfig", which of course doesn't change anything - praying for miracles.

pschatzmann, 2021-12-25:
The method has been renamed to setPinConfig().
I have corrected the example...

bazworkshops, 2021-12-25:
Yeah, I kinda realized that, but any ideas as to why it's not making sound? Like I said, the old clarinet-i2s works with the same i2s pins that I am using in InstrumentI2S. And Merry Christmas, if that's your thing!

pschatzmann, 2021-12-25:
That's strange. I am not aware of any change that might cause this. 
Double check that you call the begin method on the output and that the I2S pins are really right.

I also noticed that the default i2s config had some strange values and the output was going to the internal DAC. I corrected this one as well.

I also noticed that the sound that is produced by the example is pretty bad - so I adjusted the sketch a bit. Unfortunatly I could only test with an AudioKit, and not with the setup of the example sketch yet...

bazworkshops, 2021-12-25:
With a change of I2SStream to ArdI2SOut the latest InstrumentToI2S works! Yay! amplitude = 128 clips like crazy, but seems ok at 64. That might just be a mismatch between my amp and speaker tho. Thanks for all your work on this!

pschatzmann, 2021-12-25:
Thanks for your feed back. 128 is really the max value and I agree that it is a bad idea to use this.
I have adjusted the examples...

closed 2022-01-27 by pschatzmann