pschatzmann/arduino-stk #5: Not an issue Maybe we can have a equation composer?

NoNamedCat, 2022-10-13:
Man TKS A LOT for this library. I hope I can learn to use it in my spare time.
Just a question...
Is very difficult to add a tool like this one?
[https://esptiny86.github.io/espnode86](url)
This tool is a composer (graphical node patcher) for the esptiny86 project (Another synth library but for the ESP8266):
[https://github.com/esptiny86/espsynth86](url)

pschatzmann, 2022-10-14:
I had a quick look at [espsynth86](https://github.com/pschatzmann/arduino-stk/issues/url) and it seems that it has an API where you can pull indicividual samples.  You could work directly with this, but I would recommend that you create some simple interface class like I have done for Maximilian, STK or Faust or Mozzi

NoNamedCat, 2022-10-14:
Do you look at the espnode86? the graphical tool for making the patches?