pschatzmann/arduino-stk #10: fatal error: pgmspace.h: No such file or directory  #  include "pgmspace.h"

poetaster, 2024-01-24:
I downgraded the nano boards avr version to 1.8.2 but still get this error.  I haven't tried on the pi pico yet, but that would also be an option, or?

Thanks and regards,
Mark

pschatzmann, 2024-01-24:
Depending on what you want to do, the RP2040 might be an option. The best performance however and the least restrictions  you will get with an ESP32 or any of its variants...

E.g. the Pico has no floating point processor, does not support duplex I2S, has no internal DAC and only some limited ADCs

closed 2024-01-24 by pschatzmann

poetaster, 2024-01-24:
Ok, thanks very much. At this rate, I'll either stick to the Ksoloti boards or get some daisys. The ESPs don't have enough ADC (pico neither) for some cases, even if I switch a number of inputs to digital. Hmmm. 

Thanks again for all your efforts!

Mark