pschatzmann/arduino-stk #11: multiple definition of `__sync_synchronize';

poetaster, 2024-02-01:
I'm not sure if I'm doing this incorrecty, but after finally getting PWM with the VS1053Driver working I wanted to finally try the audio-tools arduino-stk library. I used an audiokit example as a basis on the RP2040 with a VS1053b. I've tried three more or less simple variation but all throw:
```
/home/mwa/.arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_238617/libraries/arduino-stk/ArdConfig.cpp.o: in function `__sync_synchronize':
/home/mwa/Arduino/libraries/arduino-stk/src/ArdConfig.cpp:9: multiple definition of `__sync_synchronize'; /tmp/arduino_build_238617/libraries/arduino-audio-tools-0.9.6/AudioTools/AudioRuntime.cpp.o:/home/mwa/Arduino/libraries/arduino-audio-tools-0.9.6/src/AudioTools/AudioRuntime.cpp:15: first defined here
collect2: error: ld returned 1 exit status
```

The sketch uses the streams-stk_generator-audiokit as the jumping off point:

```
#include "AudioTools.h"
#include "AudioLibs/AudioSTK.h"
#include "AudioLibs/VS1053Stream.h"

uint16_t sample_rate=44100;
uint8_t channels = 2;                                      // The stream will have 2 channels 
uint8_t bits_per_sample = 16;                              // 2 bytes 

Clarinet clarinet(440); // the stk clarinet instrument
STKGenerator<Instrmnt, int16_t> generator(clarinet);    // subclass of SoundGenerator
GeneratedSoundStream<int16_t> in(generator);  // Stream generated from sine wave
VS1053Stream out;    // VS1053 output
StreamCopy copier(out, in); // copy stkStream to a2dpStream
MusicalNotes notes; // notes with frequencies

// Arduino Setup
void setup(void) {
    Serial.begin(115200);
    AudioLogger::instance().begin(Serial, AudioLogger::Error);
    auto cfg = out.defaultConfig();
    cfg.sample_rate = sample_rate;
    cfg.channels = channels;
    cfg.bits_per_sample = bits_per_sample;
  
    // start Output
    Serial.println("starting Analog Output...");
    out.begin(cfg);

    // start STK input with default configuration
    Serial.println("starting STK...");
    auto cfgSTK = in.defaultConfig();
    cfgSTK.channels = 2;
    in.setNotifyAudioChange(out);
    in.begin(cfgSTK);

}

void playNoteStk() {
    static unsigned long timeout=0;
    static int note = 60;
    static bool isNoteOn = false;

    if (!isNoteOn) {
      if (millis()>timeout) {
          // play note for 1000 ms
          note += rand() % 10 - 5; // generate some random offset
          float frequency = notes.frequency(note);
          clarinet.noteOn( frequency, 0.5 );
  
          // set timeout
          timeout = millis()+1000;
      }
    } else {
      if (millis()>timeout) {
        // switch note off for 200 ms 
        clarinet.noteOff(0.5);
        
        // set timeout
        timeout = millis()+200; 
      }           
    }
    isNoteOn = !isNoteOn;
    
}

// Arduino loop - copy data
void loop() {
    playNoteStk();
    copier.copy();
}
///
```

pschatzmann, 2024-02-01:
I suggest that you comment out #define FIX_SYNC_SYNCHRONIZE in the AudioConfig.h
ps. your logic in the loop will not work this way. You need to call playNoteStk() e.g. time driven or by some events because it takes many copy calls to output any sound

poetaster, 2024-02-01:
Ah, that sync function did the trick. I'm just running your examples but to the VS1053 stream. Just tried the all instruments thing which loops. Sounds, well, very interesting. But not like the STK I know (I've been using it among other things for 25+ years). 

I'll take a crack at it with I2S when I'm back from a conference. Thanks!

pschatzmann, 2024-02-01:
No worries: I was not really working with any RP2040 and did not do any extensive testing.
The STK is using floating point operations. Unfortunately this is one of the weak spots of the RP2040!
Not sure if this is the source of your issues...

poetaster, 2024-02-01:
> Unfortunately this is one of the weak spots of the RP2040!
> Not sure if this is the source of your issues...

Ah, that could well be the issue. Thanks!

pschatzmann, 2024-02-01:
The only major change that I have done is that I defined StkFloat as float because double was definitely too slow for all micro controllers