pschatzmann/arduino-stk #14: ArdI2SOut is not declared if used with an ESP32-S3

Ic-ks, 2024-06-28:
I migrated a running platform-io project (also reproducible with your [example](https://github.com/pschatzmann/Arduino-STK/wiki/Output-of-Sound#arduino-audio-tools---i2s)) from a `lolin_c3_mini` to a `esp32-s3-devkitm-1`. After I changed the platformio.ini from:
```
[env:lolin_c3_mini]
platform = espressif32
board = lolin_c3_mini
framework = arduino
lib_deps = 
	https://github.com/pschatzmann/arduino-audio-tools.git
	https://github.com/pschatzmann/Arduino-STK
...
```
to: 
```
[env:esp32-s3-devkitm-1]
platform = espressif32
board = esp32-s3-devkitm-1
framework = arduino
lib_deps = 
	https://github.com/pschatzmann/arduino-audio-tools.git
	https://github.com/pschatzmann/Arduino-STK
...
```
I got the following build errors:
```
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:11:1: error: 'ArdI2SOut' does not name a type; did you mean 'ArdCommonOut'?
 ArdI2SOut::ArdI2SOut(unsigned nChannels, i2s_port_t i2s_number)
 ^~~~~~~~~
 ArdCommonOut
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:20:6: error: 'ArdI2SOut' has not been declared
 void ArdI2SOut::setI2SConfig(i2s_config_t& i2s_config) {
      ^~~~~~~~~
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp: In function 'void stk::setI2SConfig(i2s_config_t&)':
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:21:5: error: invalid use of 'this' in non-member function
     this->i2s_config = i2s_config;
     ^~~~
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp: At global scope:
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:24:6: error: 'ArdI2SOut' has not been declared
 void ArdI2SOut::setPinConfig(i2s_pin_config_t& pin_config) {
      ^~~~~~~~~
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp: In function 'void stk::setPinConfig(i2s_pin_config_t&)':
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:25:5: error: invalid use of 'this' in non-member function
     this->pin_config = pin_config;
     ^~~~
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp: At global scope:
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:28:6: error: 'ArdI2SOut' has not been declared
 void ArdI2SOut::begin(){
      ^~~~~~~~~
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp: In function 'void stk::begin()':
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:30:5: error: 'i2s_config' was not declared in this scope
     i2s_config.bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT;
     ^~~~~~~~~~
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:30:5: note: suggested alternative: 'i2s_config_t'
     i2s_config.bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT;
     ^~~~~~~~~~
     i2s_config_t
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:34:24: error: 'i2s_number' was not declared in this scope
     i2s_driver_install(i2s_number, &i2s_config, 0, NULL);
                        ^~~~~~~~~~
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:34:24: note: suggested alternative: 'i2s_write'
     i2s_driver_install(i2s_number, &i2s_config, 0, NULL);
                        ^~~~~~~~~~
                        i2s_write
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:36:17: error: invalid use of 'this' in non-member function
     i2s_set_pin(this->i2s_number, &pin_config);
                 ^~~~
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:36:36: error: 'pin_config' was not declared in this scope
     i2s_set_pin(this->i2s_number, &pin_config);
                                    ^~~~~~~~~~
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:36:36: note: suggested alternative: 'i2s_config_t'
     i2s_set_pin(this->i2s_number, &pin_config);
                                    ^~~~~~~~~~
                                    i2s_config_t
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:37:5: error: 'active' was not declared in this scope
     active = true;
     ^~~~~~
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:37:5: note: suggested alternative: 'ctime'
     active = true;
     ^~~~~~
     ctime
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp: At global scope:
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:42:6: error: 'ArdI2SOut' has not been declared
 void ArdI2SOut::write(StkFloat value) {
      ^~~~~~~~~
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp: In function 'void stk::write(stk::StkFloat)':
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:43:9: error: 'active' was not declared in this scope
     if (active){
         ^~~~~~
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:43:9: note: suggested alternative: 'ctime'
     if (active){
         ^~~~~~
         ctime
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp: At global scope:
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:49:6: error: 'ArdI2SOut' has not been declared
 void ArdI2SOut::write(int16_t value) {
      ^~~~~~~~~
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp: In function 'void stk::write(int16_t)':
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:50:20: error: 'nChannels' was not declared in this scope
     int16_t buffer[nChannels];
                    ^~~~~~~~~
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:52:9: error: 'buffer' was not declared in this scope
         buffer[j]=value;
         ^~~~~~
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:52:9: note: suggested alternative: 'setbuffer'
         buffer[j]=value;
         ^~~~~~
         setbuffer
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:55:15: error: 'i2s_number' was not declared in this scope
     i2s_write(i2s_number, (const char *)buffer, byteCount, &written, portMAX_DELAY);
               ^~~~~~~~~~
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:55:15: note: suggested alternative: 'i2s_write'
     i2s_write(i2s_number, (const char *)buffer, byteCount, &written, portMAX_DELAY);
               ^~~~~~~~~~
               i2s_write
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:55:41: error: 'buffer' was not declared in this scope
     i2s_write(i2s_number, (const char *)buffer, byteCount, &written, portMAX_DELAY);
                                         ^~~~~~
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:55:41: note: suggested alternative: 'setbuffer'
     i2s_write(i2s_number, (const char *)buffer, byteCount, &written, portMAX_DELAY);
                                         ^~~~~~
                                         setbuffer
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:55:49: error: 'byteCount' was not declared in this scope
     i2s_write(i2s_number, (const char *)buffer, byteCount, &written, portMAX_DELAY);
                                                 ^~~~~~~~~
*** [.pio/build/esp32-s3-devkitm-1/libbd8/Arduino-STK/ArdI2SOut.cpp.o] Error 1
```
A quick look into `ArdI2SOut.h` shows that `ESP32X` is not defined if the board is a S3:
[image]

pschatzmann, 2024-06-28:
So why don't you just add a -DARDUINO_ESP32S3_DEV in the build_flags is PltfromIO is not providing the correct values

closed 2024-06-28 by pschatzmann

Ic-ks, 2024-06-28:
OMG. Apologies for this silly problem