pschatzmann/arduino-stk #15: STKLentPitShift does not implement stk::StkFloat tick(stk::StkFloat input, unsigned int channel = 0)

Ic-ks, 2024-06-28:
Sorry to bother you again and thanks for your help last time :)

After including  `AudioSTK.h` I got this build error:
```
In file included from src/main.cpp:8:
.pio/libdeps/esp32-s3-devkitm-1/audio-tools/src/AudioLibs/AudioSTK.h: In member function 'virtual audio_tools::AudioEffect* audio_tools::STKLentPitShift::clone()':
.pio/libdeps/esp32-s3-devkitm-1/audio-tools/src/AudioLibs/AudioSTK.h:276:37: error: invalid new-expression of abstract class type 'audio_tools::STKLentPitShift'
     return new STKLentPitShift(*this);
                                     ^
.pio/libdeps/esp32-s3-devkitm-1/audio-tools/src/AudioLibs/AudioSTK.h:269:7: note:   because the following virtual functions are pure within 'audio_tools::STKLentPitShift':
 class STKLentPitShift : public AudioEffect, public stk::LentPitShift {
       ^~~~~~~~~~~~~~~
In file included from .pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/Chorus.h:4,
                 from .pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/StkAll.h:16,
                 from .pio/libdeps/esp32-s3-devkitm-1/audio-tools/src/AudioLibs/AudioSTK.h:8,
                 from src/main.cpp:8:
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/Effect.h:40:20: note:   'virtual stk::StkFloat stk::Effect::tick(stk::StkFloat, unsigned int)'
   virtual StkFloat tick (StkFloat input, unsigned int channel=0)=0;
                    ^~~~
*** [.pio/build/esp32-s3-devkitm-1/src/main.cpp.o] Error 1
```

I could workaround this error by adding the missing tick function to the `STKLentPitShift` class in `AudioSTK.h`:
```
  virtual stk::StkFloat tick(stk::StkFloat input, unsigned int channel = 0) override {
    return stk::LentPitShift::tick(input);
  }
```
[image]

At least `Sitar` seams to work with this workaround. But I am not sure if this is the right approach to solve this error.

pschatzmann, 2024-06-28:
I can't reproduce your issue and I would not see why this would be necessary!

closed 2024-06-28 by pschatzmann

Ic-ks, 2024-06-29:
Based on your [example code](https://github.com/pschatzmann/Arduino-STK/wiki/Output-of-Sound#arduino-audio-tools---i2s) I created a pio project to reproduce the error: https://github.com/Ic-ks/arduino-stk-issue
The dependencies in `platform.ini` are pinned to the latest version. Could it be that a specific c++ version is needed and must be pinned by an `build_flags` argument?

pschatzmann, 2024-06-29:
Did you try with the master branches as I did ?
|-- audio-tools @ 0.9.8+sha.b070b92
|-- Arduino-STK @ 1.0.1+sha.54e17a0

Ic-ks, 2024-06-29:
I pinned it to:
```
lib_deps = 
	https://github.com/pschatzmann/arduino-audio-tools#v0.9.8
	https://github.com/pschatzmann/Arduino-STK#v1.1.2
```

If I change the version of Arduino-STK to 1.0.1 I get a very strange error which complains that the `arduino.h` is missing:

```
In file included from .pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdStreamBinaryOut.h:8,
                 from .pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/StkAll.h:115,
                 from .pio/libdeps/esp32-s3-devkitm-1/audio-tools/src/AudioLibs/AudioSTK.h:8,
                 from src/main.cpp:2:
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdStreamOutCommon.h:10:10: fatal error: arduino.h: No such file or directory
```

pschatzmann, 2024-06-29:
You would need to remove the version number to use the master branch e.g. 	https://github.com/pschatzmann/Arduino-STK instead of https://github.com/pschatzmann/Arduino-STK#v1.1.2

I created a new release which contains the latest corrections

Ic-ks, 2024-06-29:
I deleted the .pio folder to enforce a clean build. After that I removed the version numbers. The current `platform.ini`:

```
[platformio]
description = Arduino STK Issue
default_envs = esp32-s3-devkitm-1

[env:esp32-s3-devkitm-1]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32-s3-devkitm-1
framework = arduino
lib_deps = 
	https://github.com/pschatzmann/arduino-audio-tools
	https://github.com/pschatzmann/Arduino-STK
build_flags = -DARDUINO_ESP32S3_DEV -DCORE_DEBUG_LEVEL=5 -Wno-unused-variable -Wno-unused-but-set-variable -Wno-unused-function -Wno-format-extra-args
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
```

Now I get this error:
```
Compiling .pio/build/esp32-s3-devkitm-1/lib4f9/Arduino-STK/BeeThree.cpp.o
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:11:1: error: 'ArdI2SOut' does not name a type; did you mean 'ArdCommonOut'?
```

Ic-ks, 2024-06-29:
If I change the board to `esp32dev` and remove the `-DARDUINO_ESP32S3_DEV` flag then I get a successful build. With the `esp32-s3-devkitm-1` board the `ArdI2Out` error appears:
```

Compiling .pio/build/esp32-s3-devkitm-1/lib4f9/Arduino-STK/BeeThree.cpp.o
.pio/libdeps/esp32-s3-devkitm-1/Arduino-STK/src/ArdI2SOut.cpp:11:1: error: 'ArdI2SOut' does not name a type; did you mean 'ArdCommonOut'?
 ArdI2SOut::ArdI2SOut(unsigned nChannels, i2s_port_t i2s_number) 
```
The error occurs with and without the `-DARDUINO_ESP32S3_DEV` flag

pschatzmann, 2024-06-29:
I deleted the obsolete ArdI2SOut.h and ArdI2SOut.cpp: I think it does not make any sense to keep this old implementation that only works for the old ESP32 versions.

This should resolve your issues

Ic-ks, 2024-06-29:
Nice, it works. Thank you very much for your incredible support!